mprograms/QMC5883LCompass #5: Need help with applying calibration data

gone-bush, 2020-07-09:
I am trying to convert your QMC5883LCompass.cpp calibration code to run in a multi-threaded application on a Raspberry Pi. The problem I am having is the use of the calibration data.

To illustrate by way of an example - and I'll restrict it to the x axis for simplicity.

Let xMin = 102 and xMax = 108, this gives xOffset = 105, xDelta = 3 and xScale = 1.

Now assume that xRaw = 103. Therefore xRaw - xOffset = -2 and then xCalibrated = -2!!!!.

What have I misunderstood?

TIA,
Kevin Gilbert

TheNitek, 2020-07-19:
The claibration is explained quite well here: https://appelsiini.net/2018/calibrate-magnetometer/

gone-bush, 2020-07-19:
Thank you for your reply. I have read the article but it doesn't answer my question. Let me illustrate:

Let max(x) = min(x) + 10, ditto for the other axes. This then gives:

offset_x = (min(x) + 10 + min(x)) / 2 = min(x) + 5

If we let sensor_x = min(x) + 2 we therefore get

corrected_x = min(x) + 2 - (min(x) + 5)) = -3.

Continuing,

avg_delta_x = (min(x) + 10 - min(x)) / 2 = 5

and, according the above assumption that the other axes are the same, we get

avg_delta = avg_delta_x

and

scale_x = 1

hence

corrected_x = (min(x) + 2 - (min(x) + 5)) * 1 = -3

What am I missing?
Kevin

TheNitek, 2020-07-20:
You did a lot of math now, but you did not describe your problem/what confuses you. Why don‘t you like the values the calibration gives you? If sensor only provides you data in a range out of 10 values, then yes the nirmalized data will be between -5 and 5

gone-bush, 2020-07-20:
My problem is that corrected_x is a constant, -3 in my example.

If min(x) = 100, the corrected_x = -3. If min(x) = 200, then corrected_x = -3.

TheNitek, 2020-07-20:
That‘s why you normalize data. It‘s moving the  the same relative values to a specific absolute value. Maybe take a closer look at the plots in original artical and look how the point clouds get moved around

gone-bush, 2020-07-20:
OK, I'm beginning to understand. I'll take a closer look at those graphs. I was expecting the corrected number to be approximately equal to the raw number. WRONG! :(

closed 2021-12-23 by mprograms