mprograms/QMC5883LCompass #9: solving hangup of the sensor (in continuous mode)

Stef-aap, 2021-04-24:
sometimes the sensor hangs, resetting or init doesn´t solve the problem. There are 2 ways to solve the problem:
1. powercycle the sensor
2. modify the Data_Ready method as shown below

bool QMC5883LCompass::Data_Ready () {
	Wire.beginTransmission ( _ADDR ) ;
	Wire.write ( 0x06 ) ;
	int err = Wire.endTransmission();
	if (!err) {
		Wire.requestFrom(_ADDR, (byte)1);
    uint8_t Status = Wire.read() ;
    if ( ( Status & 0x01 ) > 0 ) return true ;
    
// *********************************************************************
//SM: to prevent hangup, if skipbit DOR (0x04) is set 
// youe to perform a read to get it working again (but data will be invalid)    
if ( ( Status & 0x04 ) > 0 ) read() ; 
// *********************************************************************
   
  }
  return false ;
}

capedra, 2021-05-22:
I can't find this Data_Ready function anywhere.

ZIVsc, 2023-02-05:
where is this data_Ready function?