mprograms/QMC5883LCompass #13: How to detect a car around the sensor

SomeDevWeb, 2022-05-06:
Hi. 
This is not an issue, but an advice request or an example request.

Firstly, this library is quite awesome and I can make it work with the QMC5883L sensor, but I need an action that does not exist in any example, as I see. Maybe I'm wrong, cause I'm still new in the Arduino code world, so please excuse me if there is already there.

So, what I need is to be able to **detect the presence of a car above the sensor**. 
I am working on an application that basically detects a car in front of the gate and triggers the gate to open, so the driver doesn't have to get out and open the gate. I am also new in the magnetometry field, but I researched a bit and it seems that you can detect the car by determining the change that it triggers in the magnetic field around the sensor.
Is there an example of that or can you possibly give some advice in that direction?
Thank you very much for your time and support!

frollard, 2023-02-21:
It might be tricky with earth's magnetic field...You might need to place a strong magnet nearby (also under the car) and get a known value for the field with no car, any disturbance to the field could be considered a car. (imagine a bar magnet standing on end, all the field lines would go up in the air, outward, down, through the dirt, and loop into the other pole. When there's a car, the lines may couple more strongly with the metal than the air, and form a tighter smaller loop than when there's only air to conduct the field. The angle of the field would change with metal. Capture a few examples, plot the data, and build an algorithm around it.

ZIVsc, 2023-02-21:
Hey Fro, do u work on iot projects at all?

frollard, 2023-02-21:
> Hey Fro, do u work on iot projects at all?

Nope.

closed 2023-08-03 by mprograms