mprograms/QMC5883LCompass #14: Filtering the Influence of Surrounding Objects

Hs-Cheng, 2022-08-23:
Hi,

Thank you for writing this useful library.

When I am operating the compass, I noticed that the readings of the compass is vulnerable to surrounding objects. The readings are very different when I put the compass on my table and 30 centimeters above the table, while keeping the compass positioning to the same direction. Also, when I move the compass to different places (and the same height) on my table, the readings are also quite different. 

I am already doing the calibration and reading smoothing. Do you think there is a way to filter out the influence of surrounding objects, and make the compass only respond to the magnetic field of the Earth?

Thank you for your assistance.

All the best,
HC

frollard, 2023-02-21:
https://www.best-microcontroller-projects.com/hmc5883l.html
Realistically, it's just down to calibration. The sensor can't tell the difference between earth's magnetic field undisturbed, versus earth's magnetic field threading itself in/around the metal of your desk. The field lines are just in different directions. At best you can calibrate for a single location, and use it in that location. Anywhere you move after calibrating will need a new calibration. There may be a way to use additional sensors to detect your desk and apply a 'known near desk' calibration when near the desk, but really it's just a limitation of magnetic fields.
Also possible to use a 9 axis accel/gyro/mag sensor and infer from one to determine the others. Gravity is down, so you can know the normal axis from the accelerometer z, if magnet isn't moving, you can zero out the gyros, etc. It's still dead reckoning because there is no ground truth but you can infer some stability from different types of sensors.

ZIVsc, 2023-02-21:
I'm currently implementing this sensor with a water meter. To read the
magnetic field the spinning magnet to detect flow. Having issues with noise
at no flow times.

frollard, 2023-02-21:
Do you have access to install a magnetic rotary encoder on the axle of the meter?
https://www.aliexpress.com/item/1005002339509627.html
AS5040-ASST AS5040 magnetic encoder, etc. They are basically the same but different, intentionally converting a rotating field into a position, instead of just using the 3axis as a glorified hall effect switch. 
Otherwise, if you're getting noise, you have to find a way to algorithmically detect full rotations versus just jittering readings. If the magnetic field is jittering full swing positive to negative then this sensor simply won't work for you.

closed 2023-08-03 by mprograms