mprograms/QMC5883LCompass #21: The  i2c scanner don’t find QMC5883LCompass and it don’t work on ESP32

mateusmsantin, 2023-07-24:
Hi! 
I used this code to find QMC5883L Compass, but don’t found. 
The code compile on ESP32, but don’t show data. 
What can I do to test it ?

`#include <Wire.h>

// Set I2C bus to use: Wire, Wire1, etc.
#define WIRE Wire

#define SDA 14 //3 // para teste 14
#define SCL 2 // para teste 2

void setup() {
  Wire.begin(SDA, SCL); 
  //Wire.begin(); 

  Serial.begin(9600);
  while (!Serial)
     delay(10);
  Serial.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    WIRE.beginTransmission(address);
    error = WIRE.endTransmission();
    Serial.print (" # ");
    Serial.print (address);
    if (error == 0)
    {
      Serial.print("\n I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknown error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}`

mprograms, 2023-08-03:
If you are not finding the chip's address then it is a hardware issue, not an issue with the library.

There are a few possibilities if you are not finding the chip.

- Your wiring might be wrong.
- The chip is bad.
- The chip is not QMC5883L.
- There is an issue with the I2C bus. In this case, try a different Arduino processor.

mateusmsantin, 2023-10-09:
The library work with **NodeMcu**  ? 
When I connected the pins **SCL** - _D1_ / **SDA** - _D2_ and try calibrate, the program stop to work. 

21:58:00.329 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
21:58:00.363 -> H!��1�D��LH�l$hr�9�This will provide calibration settings for your QMC5883L chip. When prompted, move the magnetometer in all directions until the calibration is complete.
21:58:00.590 -> Calibration will begin in 5 seconds.
21:58:05.529 -> CALIBRATING. Keep moving your sensor...
21:58:08.456 -> 
21:58:08.456 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
21:58:08.519 -> 
21:58:08.519 -> Soft WDT reset
21:58:08.519 -> 
21:58:08.519 -> Exception (4):
21:58:08.583 -> epc1=0x40201618 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
21:58:08.649 -> 
21:58:08.649 -> >>>stack>>>
21:58:08.682 -> 
21:58:08.682 -> ctx: cont
21:58:08.682 -> sp: 3ffffdf0 end: 3fffffd0 offset: 0160
21:58:08.714 -> 3fffff50:  3fffdad0 00001388 00000085 40203904  
21:58:08.781 -> 3fffff60:  0000140d 00000000 00000000 00000000  
21:58:08.846 -> 3fffff70:  00000000 00000000 3ffee8d0 3ffee950  
21:58:08.880 -> 3fffff80:  3fffdad0 00000027 3ffee8d0 3ffee950  
21:58:08.945 -> 3fffff90:  3fffdad0 3ffee688 3ffee8d0 4020109d  
21:58:08.978 -> 3fffffa0:  feefeffe feefeffe feefeffe 3ffee950  
21:58:09.044 -> 3fffffb0:  3fffdad0 00000000 3ffee924 40202228  
21:58:09.078 -> 3fffffc0:  feefeffe feefeffe 3fffdab0 40100f91  
21:58:09.144 -> <<<stack<<<

blopa1961, 2024-04-19:
You might want to take a look at the Compass I published here:
https://www.rcgroups.com/forums/showthread.php?4535205-S-PORT-Compass-for-FrSky-telemetry

Working Arduino Nano code included, should be easily ported to nodeMCU.