mprograms/QMC5883LCompass #25: error: narrowing conversion of '65000' from 'long int' to 'int'

whyameye, 2023-08-12:
When trying to compile using platformIO and LGT8328p I get the following error:
`.pio/libdeps/program_via_ArduinoISP/QMC5883LCompass/src/QMC5883LCompass.cpp:152:80: error: narrowing conversion of '65000' from 'long int' to 'int' inside { } [-Wnarrowing]`

I changed line 152 of `QMC5883LCompass.cpp` from
`int calibrationData[3][2] = {{65000, -65000}, {65000, -65000}, {65000, -65000}};`
to
`int calibrationData[3][2] = {{32767, -32768}, {32767, -32768}, {32767, -32768}};`

and that solved the error and seemed to give the correct results in my test code. I haven't studied the code so I don't know if this is a viable solution.

virtyaluk, 2023-08-15:
Hey @whyameye,

I'm pretty sure on lgt8f328p an int stores a 16-bit (2-byte) value. A 2-byte signed integer takes a minimum value of -32,768 to the maximum value of 32,767. That's why you get that error. 

You may want to check other numeric types yourself:

```c
#define PRINT_SIZE_OF(type) Serial.println("Size of " str(type) " is: " + String(sizeof(type)*8) + " bits");

void loop() {   
    PRINT_SIZE_OF(int);
    PRINT_SIZE_OF(int32_t);  
    PRINT_SIZE_OF(uint32_t);  
    PRINT_SIZE_OF(int_least32_t);  
    PRINT_SIZE_OF(uint_least32_t);  
    PRINT_SIZE_OF(int64_t);  
    PRINT_SIZE_OF(uint64_t);  
    delay(1000);
} 
```

The proposed code change may introduce inconsistencies in the calibration algorithm which is already super unstable (tested with 4 different QMC5883 chips) and doesn't work on some of the MCU boards (NodeMCU, for example) without manual code updates. For the sake of simplicity, you could just go with long int instead of int.

Thanks.

BenjaminDerProgrammierer, 2023-08-15:
I think I've solved it:
put 
`int calibrationData[3][2] = {{(int)65000, (int)-65000}, {(int)65000, (int)-65000}, {(int)65000, (int)-65000}};`
instead of
`int calibrationData[3][2] = {{65000, -65000}, {65000, -65000}, {65000, -65000}};`

whyameye, 2023-08-15:
I'm confused. This library is intended I believe primarily for the Arduino? And the Arduino datatype for int is 2 bytes making its min and max value -32768 to 32767:
https://www.arduino.cc/reference/en/language/variables/data-types/int/

so regardless of the MCU used I feel like assigning an int -65000 or 65000 in the Arduino framework would never make sense and should at the least flag a warning by the compiler in all cases. Where is the flaw in my logic and if my logic isn't flawed I think the best we can do if we keep the datatype int is my original proposal. If we want to change to long int then we need to change it wherever these values might be referenced in other variables, not in just the one line.

mprograms, 2023-08-16:
@whyameye , you are correct.

Version 1.2.x included some updates to the calibration functions. Version 1.2.1 added default values for the to the calibrate() function. Unfortunately, the type was not updated and should have been changed to long.

A new branch v1.2.2 has been created. Before I send a new release, let us know if this branch fixes your issue.

whyameye, 2023-08-17:
yes branch v1.2.2 fixes it! :) Thanks for this.

closed 2023-08-19 by mprograms